pixelmatix/SmartMatrix #160: ESP32 call to draw the ellipse function is bound to die problem

FettLuo, 2021-11-28:
**Hardware:** ESP32-WROOM-32 (4MB)
**Software:** Arduino+FeatureDemo example

**Situation:**
Under arduino, up to a few thousand times, it is bound to die.
Under esp-idf, a fixed number of times, inevitably dead.(I made a transplant version)

**Code** (add to _void loop()_, resolution keep the default can also be):
```
    static int count=0;
    int x0, y0, x1, y1, x2, y2, radius, radius2;
    x0 = random(matrix.getScreenWidth());
    y0 = random(matrix.getScreenHeight());
    radius = random(matrix.getScreenWidth());
    radius2 = random(matrix.getScreenWidth());
    rgb24 fillColor = {(uint8_t)random(192), (uint8_t)random(192), (uint8_t)random(192)};
    rgb24 outlineColor = {(uint8_t)random(192), (uint8_t)random(192), (uint8_t)random(192)};
    backgroundLayer.drawEllipse(x0, y0, radius, radius2, outlineColor);
    count++;
    printf("Draw time:%d count:%d\n", millis(), count);
    return;
```

Is there something I did wrong somewhere, even with the default demo, running for a while will crash, in my 64*64 setting.

embedded-creations, 2021-11-28:
I don’t have time to look into this now, but there was a bug in some other drawing code that was using too small of a variable to store the product of two numbers, and the number overflowed. You might look at the drawelipse code for any of those issues, or just change all the variables used in that code to larger types and see if that fixes it

FettLuo, 2021-11-29:
Thanks, I'll try to fix it

FettLuo, 2021-12-02:
When the random number appears to be zero, the radius passed to drawEllipse, it will dead-loop.
It's simple though and took a long time to find 😄

embedded-creations, 2021-12-02:
Thanks for finding that, and sorry it took a long time to find!

embedded-creations, 2021-12-02:
Would adding this to the top of drawEllipse() fix the issue?

```
    // radius must be > 0
    if(!radiusX || !radiusY)
        return;
```

FettLuo, 2021-12-03:
> Would adding this to the top of drawEllipse() fix the issue?
> 
> ```
>     // radius must be > 0
>     if(!radiusX || !radiusY)
>         return;
> ```

Yes, very stable now

closed 2022-01-05 by FettLuo

embedded-creations, 2022-01-05:
I haven't committed your fix to the repo yet, but it's in my local copy, ready to commit when I switch back to working on SmartMatrix Library again.  Thanks for your help!