pixelmatix/SmartMatrix #165: Update to arduino-esp32 v2.0.3

Lukaswnd, 2022-05-06:
hi, due to some changes in the arduino-esp32 directories and including paths, Smartmatrix is currently not compatible with [arduino-esp32](https://github.com/espressif/arduino-esp32) v2.0.0 or greater and thereby with platformio: platform-espressif32 (arduino framework) v4.0.0. 
I found the following fork, which fixes the issue. Im not that great with git, but somebody else might create a pull request.
The fork is only 4 commits ahead, so it shouldn't be that hard.

https://github.com/porkyneal/SmartMatrix

sprior, 2022-06-06:
Thanks for this fix, I started out trying to compile the code today for the first time in months and was stuck, this got me going again.

vinhill, 2022-07-13:
For those stumbling upon this, the following has to be added to

SmartMatrix/src/esp32_i2s_parallel.c
```
#include "soc/gpio_periph.h"
#include "esp32-hal.h"
#include "hal/gpio_types.h"
```

HorstBaerbel, 2022-07-18:
I can acknowledge that [this change](https://github.com/pixelmatix/SmartMatrix/commit/011f0360de2153131b0f960211720f3c4b627518) in particular fixes the problem.