pixelmatix/SmartMatrix #166: Weird /4 scan HUB75 matrix

mansj, 2022-06-08:
I've got some very bright outdoor matrices I'd like to use with my new ESP32 hardware. They are stated as 1/4 scan from the manufacturer ([https://www.aliexpress.com/item/32671579710.html](https://www.aliexpress.com/item/32671579710.html))

None of the mappers included seem to work: [see film I shot](https://www.dropbox.com/s/646g7p6sf7tgleu/IMG_5147.MOV?dl=0) using SM_PANELTYPE_HUB75_8ROW_MOD4SCAN (v2/v3/v4 give similar outputs)

I think that the testing sketch should give me just one dot moving across the matrix, but this is giving me a row of eight somehow shrinking?

embedded-creations, 2022-06-08:
it's possible that your panel uses a chipset that isn't currently supported.  Can you identify any chip numbers from the back of the panel?

mansj, 2022-06-09:
The only "large" chip I can see says CHDG1H.1 and A9881M but I can't find much on google for it. 

I have managed to use them before with a Python library on a Raspberry Pi, but I can't find much info in the code there besides that they were set to "checkered" multiplexing?

mansj, 2022-06-09:
I heard from the seller on Aliexpress now, they say the chip is "Kinglight" if that makes any sens?