pixelmatix/SmartMatrix #171: Can't compile to ESP32

Fourtyy, 2023-04-14:
In file included from C:\Users\WINDOWS 10\AppData\Local\Temp\.arduinoIDE-unsaved2023314-1692-ei49w1.wx7l9\FeatureDemo\FeatureDemo.ino:20:
c:\Users\WINDOWS 10\OneDrive\Documents\Arduino\libraries\SmartMatrix\src/MatrixHardware_ESP32_V0.h:208:21: note: #pragma message: MatrixHardware: ESP32 forum wiring
     #pragma message "MatrixHardware: ESP32 forum wiring"
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\WINDOWS 10\AppData\Local\Temp\.arduinoIDE-unsaved2023314-1692-ei49w1.wx7l9\FeatureDemo\FeatureDemo.ino:22:
c:\Users\WINDOWS 10\OneDrive\Documents\Arduino\libraries\SmartMatrix\src/SmartMatrix.h:40:21: note: #pragma message: SmartMatrix Library version 4.000.003
 #    pragma message "SmartMatrix Library version 4.000.003"
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\WINDOWS 10\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7/tools/sdk/esp32/include/soc/esp32/include/soc/soc.h:19,
                 from C:\Users\WINDOWS 10\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7/tools/sdk/esp32/include/soc/include/soc/soc_memory_types.h:20,
                 from C:\Users\WINDOWS 10\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7/tools/sdk/esp32/include/esp_hw_support/include/soc/compare_set.h:12,
                 from C:\Users\WINDOWS 10\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7/tools/sdk/esp32/include/esp_hw_support/include/soc/spinlock.h:13,
                 from C:\Users\WINDOWS 10\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/portmacro.h:42,
                 from C:\Users\WINDOWS 10\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7/tools/sdk/esp32/include/freertos/include/freertos/portable.h:51,
                 from C:\Users\WINDOWS 10\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOS.h:63,
                 from c:\Users\WINDOWS 10\OneDrive\Documents\Arduino\libraries\SmartMatrix\src\esp32_i2s_parallel.c:22:
c:\Users\WINDOWS 10\OneDrive\Documents\Arduino\libraries\SmartMatrix\src\esp32_i2s_parallel.c: In function 'gpio_setup_out':
c:\Users\WINDOWS 10\OneDrive\Documents\Arduino\libraries\SmartMatrix\src\esp32_i2s_parallel.c:129:21: error: 'GPIO_PIN_MUX_REG' undeclared (first use in this function)
     PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[gpio], PIN_FUNC_GPIO);
                     ^~~~~~~~~~~~~~~~
C:\Users\WINDOWS 10\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7/tools/sdk/esp32/include/esp_common/include/esp_assert.h:29:74: note: in definition of macro 'TRY_STATIC_ASSERT'
             ESP_STATIC_ASSERT(__builtin_choose_expr(__builtin_constant_p(CONDITION), (CONDITION), 1), #MSG);   \
                                                                          ^~~~~~~~~
C:\Users\WINDOWS 10\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7/tools/sdk/esp32/include/soc/esp32/include/soc/soc.h:96:57: note: in expansion of macro 'IS_DPORT_REG'
 #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
                                                         ^~~~~~~~~~~~
C:\Users\WINDOWS 10\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7/tools/sdk/esp32/include/soc/esp32/include/soc/soc.h:145:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((_r), REG_SET_FIELD);                                                                  \
             ^~~~~~~~~~~~~~~~~~~
C:\Users\WINDOWS 10\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7/tools/sdk/esp32/include/soc/esp32/include/soc/io_mux_reg.h:91:46: note: in expansion of macro 'REG_SET_FIELD'
 #define PIN_FUNC_SELECT(PIN_NAME, FUNC)      REG_SET_FIELD(PIN_NAME, MCU_SEL, FUNC)
                                              ^~~~~~~~~~~~~
c:\Users\WINDOWS 10\OneDrive\Documents\Arduino\libraries\SmartMatrix\src\esp32_i2s_parallel.c:129:5: note: in expansion of macro 'PIN_FUNC_SELECT'
     PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[gpio], PIN_FUNC_GPIO);
     ^~~~~~~~~~~~~~~
c:\Users\WINDOWS 10\OneDrive\Documents\Arduino\libraries\SmartMatrix\src\esp32_i2s_parallel.c:129:21: note: each undeclared identifier is reported only once for each function it appears in
     PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[gpio], PIN_FUNC_GPIO);
                     ^~~~~~~~~~~~~~~~
C:\Users\WINDOWS 10\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7/tools/sdk/esp32/include/esp_common/include/esp_assert.h:29:74: note: in definition of macro 'TRY_STATIC_ASSERT'
             ESP_STATIC_ASSERT(__builtin_choose_expr(__builtin_constant_p(CONDITION), (CONDITION), 1), #MSG);   \
                                                                          ^~~~~~~~~
C:\Users\WINDOWS 10\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7/tools/sdk/esp32/include/soc/esp32/include/soc/soc.h:96:57: note: in expansion of macro 'IS_DPORT_REG'
 #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
                                                         ^~~~~~~~~~~~
C:\Users\WINDOWS 10\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7/tools/sdk/esp32/include/soc/esp32/include/soc/soc.h:145:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((_r), REG_SET_FIELD);                                                                  \
             ^~~~~~~~~~~~~~~~~~~
C:\Users\WINDOWS 10\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7/tools/sdk/esp32/include/soc/esp32/include/soc/io_mux_reg.h:91:46: note: in expansion of macro 'REG_SET_FIELD'
 #define PIN_FUNC_SELECT(PIN_NAME, FUNC)      REG_SET_FIELD(PIN_NAME, MCU_SEL, FUNC)
                                              ^~~~~~~~~~~~~
c:\Users\WINDOWS 10\OneDrive\Documents\Arduino\libraries\SmartMatrix\src\esp32_i2s_parallel.c:129:5: note: in expansion of macro 'PIN_FUNC_SELECT'
     PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[gpio], PIN_FUNC_GPIO);
     ^~~~~~~~~~~~~~~
C:\Users\WINDOWS 10\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7/tools/sdk/esp32/include/esp_common/include/esp_assert.h:29:31: error: first argument to '__builtin_choose_expr' not a constant
             ESP_STATIC_ASSERT(__builtin_choose_expr(__builtin_constant_p(CONDITION), (CONDITION), 1), #MSG);   \
                               ^~~~~~~~~~~~~~~~~~~~~
C:\Users\WINDOWS 10\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7/tools/sdk/esp32/include/soc/esp32/include/soc/soc.h:96:38: note: in expansion of macro 'TRY_STATIC_ASSERT'
 #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
                                      ^~~~~~~~~~~~~~~~~
C:\Users\WINDOWS 10\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7/tools/sdk/esp32/include/soc/esp32/include/soc/soc.h:145:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((_r), REG_SET_FIELD);                                                                  \
             ^~~~~~~~~~~~~~~~~~~
C:\Users\WINDOWS 10\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7/tools/sdk/esp32/include/soc/esp32/include/soc/io_mux_reg.h:91:46: note: in expansion of macro 'REG_SET_FIELD'
 #define PIN_FUNC_SELECT(PIN_NAME, FUNC)      REG_SET_FIELD(PIN_NAME, MCU_SEL, FUNC)
                                              ^~~~~~~~~~~~~
c:\Users\WINDOWS 10\OneDrive\Documents\Arduino\libraries\SmartMatrix\src\esp32_i2s_parallel.c:129:5: note: in expansion of macro 'PIN_FUNC_SELECT'
     PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[gpio], PIN_FUNC_GPIO);
     ^~~~~~~~~~~~~~~
C:\Users\WINDOWS 10\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7/tools/sdk/esp32/include/esp_common/include/esp_assert.h:29:31: error: expression in static assertion is not an integer
             ESP_STATIC_ASSERT(__builtin_choose_expr(__builtin_constant_p(CONDITION), (CONDITION), 1), #MSG);   \
[... 254 lines cut ...]
                 I2S_OUT_RST
c:\Users\WINDOWS 10\OneDrive\Documents\Arduino\libraries\SmartMatrix\src\esp32_i2s_parallel.c:282:27: error: 'I2S1O_DATA_OUT0_IDX' undeclared (first use in this function); did you mean 'I2S_TX_DATA_NUM_S'?
             sig_data_base=I2S1O_DATA_OUT0_IDX;
                           ^~~~~~~~~~~~~~~~~~~
[... cut ...]
c:\Users\WINDOWS 10\OneDrive\Documents\Arduino\libraries\SmartMatrix\src\esp32_i2s_parallel.c:384:69: error: 'ESP_INTR_FLAG_LEVEL1' undeclared (first use in this function); did you mean 'ESP_ERR_FLASH_BASE'?
     esp_intr_alloc(ETS_I2S1_INTR_SOURCE, (int)(ESP_INTR_FLAG_IRAM | ESP_INTR_FLAG_LEVEL1), i2s_isr, NULL, NULL);
                                                                     ^~~~~~~~~~~~~~~~~~~~
                                                                     ESP_ERR_FLASH_BASE

exit status 1

Compilation error: exit status 1

Karpov03, 2023-07-05:
Include the below lines in SmartMatrix/src/esp32_i2s_parallel.c

#include "soc/gpio_periph.h"
#include "esp32-hal.h"
#include "hal/gpio_types.h"

netmindz, 2023-10-25:
PR created https://github.com/pixelmatix/SmartMatrix/pull/175

davemaster, 2024-06-26:
> Include the below lines in SmartMatrix/src/esp32_i2s_parallel.c
> 
> #include "soc/gpio_periph.h" 
> #include "esp32-hal.h" 
> #include "hal/gpio_types.h"

that fix, works

this:

```
#ifndef GPIO_PIN_MUX_REG
#include "soc/gpio_periph.h"
#include "esp32-hal.h"
#include "hal/gpio_types.h"
#endif
```

[image]