pixelmatix/SmartMatrix #172: Can't seem to increase brightness.

zhanex3, 2023-05-09:
Using a Teensy 4 and SmartLED V5.  

Seems my brightness caps at around a power draw of 0.08 amps.  I've tried lowering the kRefreshDepth, refresh rate, and buffer but the brightness remains the same level when set at 255. 

Every now and then when i'm uploading a new program, the leds "glitch" and get really bright, I can get power draw of .11 - 1 amp depending on my pixels are turned on during the glitches.

embedded-creations, 2023-05-09:
This is a better topic for the forum at community.pixelmatix.com

How are you powering the panels?

when not being refreshed (during the glitch) the LEDs are on 100% of the time, vs 1/16th of the time when being refreshed so they will be much brighter

P3 LEDs are smaller and less bright than P6 for example.

zhanex3, 2023-06-30:
I see, maybe i'd need a new feature to just draw an image statically without refreshing. Even at level 3 (lowest) its not getting any bright. Thanks for your help.

closed 2023-06-30 by zhanex3