pixelmatix/SmartMatrix #183: Wy i can't compile ?! Thanks for help

Jonahan06340, 2024-06-24:
Good morning,
I'm trying to compile the example "FeatureDemo.ino" project which I put: "#define GPIOPINOUT ESP32_FORUM_PINOUT".
I selected in the arduino IDE the "DOIT ESP3 DEVKIT V1" card which actually corresponds to my "ESP-WROOM-32" card.

I also included the below lines in SmartMatrix/src/esp32_i2s_parallel.c
#include "soc/gpio_periph.h"
#include "esp32-hal.h"
#include "hal/gpio_types.h"

As you can see in the lines below, it doesn't compile!
I would be very grateful if you could direct me on this problem.
Thank you

`FQBN: esp32:esp32:esp32doit-devkit-v1
Using board 'esp32doit-devkit-v1' from platform in folder: Z:\Users\Jonathan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1
Using core 'esp32' from platform in folder: Z:\Users\Jonathan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1

cmd /c if exist "Z:\\Users\\Jonathan\\Desktop\\Projets\\libraries\\SmartMatrix\\examples\\FeatureDemo\\partitions.csv" COPY /y "Z:\\Users\\Jonathan\\Desktop\\Projets\\libraries\\SmartMatrix\\examples\\FeatureDemo\\partitions.csv" "Z:\\Users\\Jonathan\\AppData\\Local\\Temp\\arduino\\sketches\\B4DD9C163FA908B4FF969381FD486CF3\\partitions.csv"
cmd /c if not exist "Z:\\Users\\Jonathan\\AppData\\Local\\Temp\\arduino\\sketches\\B4DD9C163FA908B4FF969381FD486CF3\\partitions.csv" if exist "Z:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.1\\variants\\doitESP32devkitV1\\partitions.csv" COPY "Z:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.1\\variants\\doitESP32devkitV1\\partitions.csv" "Z:\\Users\\Jonathan\\AppData\\Local\\Temp\\arduino\\sketches\\B4DD9C163FA908B4FF969381FD486CF3\\partitions.csv"
cmd /c if not exist "Z:\\Users\\Jonathan\\AppData\\Local\\Temp\\arduino\\sketches\\B4DD9C163FA908B4FF969381FD486CF3\\partitions.csv" COPY "Z:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.1\\tools\\partitions\\default.csv" "Z:\\Users\\Jonathan\\AppData\\Local\\Temp\\arduino\\sketches\\B4DD9C163FA908B4FF969381FD486CF3\\partitions.csv"
cmd /c IF EXIST "Z:\\Users\\Jonathan\\Desktop\\Projets\\libraries\\SmartMatrix\\examples\\FeatureDemo\\bootloader.bin" ( COPY /y "Z:\\Users\\Jonathan\\Desktop\\Projets\\libraries\\SmartMatrix\\examples\\FeatureDemo\\bootloader.bin" "Z:\\Users\\Jonathan\\AppData\\Local\\Temp\\arduino\\sketches\\B4DD9C163FA908B4FF969381FD486CF3\\FeatureDemo.ino.bootloader.bin" ) ELSE ( IF EXIST "Z:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.1\\variants\\doitESP32devkitV1\\bootloader.bin" ( COPY "Z:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.1\\variants\\doitESP32devkitV1\\bootloader.bin" "Z:\\Users\\Jonathan\\AppData\\Local\\Temp\\arduino\\sketches\\B4DD9C163FA908B4FF969381FD486CF3\\FeatureDemo.ino.bootloader.bin" ) ELSE ( "Z:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esptool_py\\4.6\\esptool.exe" --chip esp32 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o "Z:\\Users\\Jonathan\\AppData\\Local\\Temp\\arduino\\sketches\\B4DD9C163FA908B4FF969381FD486CF3\\FeatureDemo.ino.bootloader.bin" "Z:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083\\esp32\\bin\\bootloader_dio_80m.elf" ) )
esptool.py v4.6
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.

...
...
...

Compilation du croquis...
"Z:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp-x32\\2302/bin/xtensa-esp32-elf-g++" -MMD -c "@Z:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083\\esp32/flags/cpp_flags" -Os -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"doitESP32devkitV1\"" -DARDUINO_PARTITION_default "-DARDUINO_HOST_OS=\"windows\"" "-DARDUINO_FQBN=\"esp32:esp32:esp32doit-devkit-v1:UploadSpeed=921600,FlashFreq=80,DebugLevel=none,EraseFlash=none\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_USB_CDC_ON_BOOT=0 "@Z:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083\\esp32/flags/defines" "-IZ:\\Users\\Jonathan\\Desktop\\Projets\\libraries\\SmartMatrix\\examples\\FeatureDemo" -iprefix "Z:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083\\esp32/include/" "@Z:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083\\esp32/flags/includes" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083\\esp32/dio_qspi/include" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.1\\cores\\esp32" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.1\\variants\\doitESP32devkitV1" "-Iz:\\Users\\Jonathan\\Desktop\\Projets\\libraries\\SmartMatrix\\src" "-Iz:\\Users\\Jonathan\\Desktop\\Projets\\libraries\\Adafruit-GFX-Library" "-Iz:\\Users\\Jonathan\\Desktop\\Projets\\libraries\\Adafruit_BusIO" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.1\\libraries\\Wire\\src" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.1\\libraries\\SPI\\src" "@Z:\\Users\\Jonathan\\AppData\\Local\\Temp\\arduino\\sketches\\B4DD9C163FA908B4FF969381FD486CF3/build_opt.h" "@Z:\\Users\\Jonathan\\AppData\\Local\\Temp\\arduino\\sketches\\B4DD9C163FA908B4FF969381FD486CF3/file_opts" "Z:\\Users\\Jonathan\\AppData\\Local\\Temp\\arduino\\sketches\\B4DD9C163FA908B4FF969381FD486CF3\\sketch\\FeatureDemo.ino.cpp" -o "Z:\\Users\\Jonathan\\AppData\\Local\\Temp\\arduino\\sketches\\B4DD9C163FA908B4FF969381FD486CF3\\sketch\\FeatureDemo.ino.cpp.o"
Utilisation du fichier déjà compilé: Z:\Users\Jonathan\AppData\Local\Temp\arduino\sketches\B4DD9C163FA908B4FF969381FD486CF3\sketch\colorwheel.c.o
In file included from Z:\Users\Jonathan\Desktop\Projets\libraries\SmartMatrix\examples\FeatureDemo\FeatureDemo.ino:35:
z:\Users\Jonathan\Desktop\Projets\libraries\SmartMatrix\src/MatrixHardware_ESP32_V0.h:208:21: note: '#pragma message: MatrixHardware: ESP32 forum wiring'
  208 |     #pragma message "MatrixHardware: ESP32 forum wiring"
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from Z:\Users\Jonathan\Desktop\Projets\libraries\SmartMatrix\examples\FeatureDemo\FeatureDemo.ino:37:
z:\Users\Jonathan\Desktop\Projets\libraries\SmartMatrix\src/SmartMatrix.h:40:21: note: '#pragma message: SmartMatrix Library version 4.000.003'
   40 | #    pragma message "SmartMatrix Library version 4.000.003"
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from z:\Users\Jonathan\Desktop\Projets\libraries\SmartMatrix\src/Layer_Scrolling.h:98,
                 from z:\Users\Jonathan\Desktop\Projets\libraries\SmartMatrix\src/SmartMatrix.h:53:
z:\Users\Jonathan\Desktop\Projets\libraries\SmartMatrix\src/Layer_Scrolling_Impl.h: In member function 'void SMLayerScrolling<RGB, optionFlags>::updateScrollingText()':
z:\Users\Jonathan\Desktop\Projets\libraries\SmartMatrix\src/Layer_Scrolling_Impl.h:273:36: warning: '--' expression of 'volatile'-qualified type is deprecated [-Wvolatile]
  273 |             if (scrollcounter > 0) scrollcounter--;
      |                                    ^~~~~~~~~~~~~
z:\Users\Jonathan\Desktop\Projets\libraries\SmartMatrix\src/Layer_Scrolling_Impl.h:281:36: warning: '--' expression of 'volatile'-qualified type is deprecated [-Wvolatile]
  281 |             if (scrollcounter > 0) scrollcounter--;
      |                                    ^~~~~~~~~~~~~
z:\Users\Jonathan\Desktop\Projets\libraries\SmartMatrix\src/Layer_Scrolling_Impl.h:289:36: warning: '--' expression of 'volatile'-qualified type is deprecated [-Wvolatile]
  289 |             if (scrollcounter > 0) scrollcounter--;
      |                                    ^~~~~~~~~~~~~
In file included from z:\Users\Jonathan\Desktop\Projets\libraries\SmartMatrix\src/Layer_Gfx_Mono.h:157,
                 from z:\Users\Jonathan\Desktop\Projets\libraries\SmartMatrix\src/SmartMatrix.h:60:
[... cut ...]
Utilisation de la bibliothèque Adafruit BusIO version 1.16.1 dans le dossier: Z:\Users\Jonathan\Desktop\Projets\libraries\Adafruit_BusIO 
Utilisation de la bibliothèque Wire version 2.0.0 dans le dossier: Z:\Users\Jonathan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\Wire 
Utilisation de la bibliothèque SPI version 2.0.0 dans le dossier: Z:\Users\Jonathan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\SPI 
exit status 1

Compilation error: exit status 1`

davemaster, 2024-06-24:
lines up of <SmartMatrix.h> line, there are many "pin definitions"; I have the same board, so, do this:

[image]

Uncomment that line, and try again

Jonahan06340, 2024-06-24:
First of all thank you. 
But I already uncommented this line in my "FeatureDemo" project !!!
And I even added this line at the top of my project: "#define GPIOPINOUT ESP32_FORUM_PINOUT".
I really don't understand where this problem comes from.

davemaster, 2024-06-25:
THEN, are You sure, You're selecting the right board? What version of ESP32 have in your Boards Manger?

Better a picture or video to diagnose the patient

davemaster, 2024-06-25:
here, your error:

299 | gpio_pad_select_gpio(CLK_PIN);

Seems like You're not defining the CLK_PIN, if You customized the ESP32-HARDWARE-V0 (or whatever file) May be You committed a mistype..

Jonahan06340, 2024-06-25:
Thank you so much.
The problem is that I didn't customize anything !

davemaster, 2024-06-26:
I have that board too. 

One more tip, about those libraries, Try (board manager) to downgrade the esp32 to 1.0.6 (yes, that old version) and compile all over again. 

[image]

Best regards

davemaster, 2024-06-26:
Also, this https://github.com/pixelmatix/SmartMatrix/pull/175 for esp32, I try this with esp32 2.0.17 and works

[image]

Jonahan06340, 2024-06-26:
Thank you very much for your help, but the problem persists!
[image]

davemaster, 2024-06-26:
But, You're trying using Arduino IDE 2.X
NO, use the 1.8.19 FIRST... try with that one.

Jonahan06340, 2024-06-26:
Finally I managed to compile the example provided in the library :)
As proposed by @davemaster  I downgraded to esp32 => 2.0.17
However, I have some Warnings that I will put for you below.

`Utilisation du fichier déjà compilé: Z:\Users\Jonathan\AppData\Local\Temp\arduino\sketches\B4DD9C163FA908B4FF969381FD486CF3\sketch\colorwheel.c.o
"Z:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\esp-2021r2-patch5-8.4.0/bin/xtensa-esp32-elf-g++" -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE "-DIDF_VER=\"v4.4.7-dirty\"" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/newlib/platform_include" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/freertos/include" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/freertos/include/esp_additions/freertos" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/freertos/port/xtensa/include" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/freertos/include/esp_additions" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/esp_hw_support/include" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/esp_hw_support/include/soc" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/esp_hw_support/include/soc/esp32" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/esp_hw_support/port/esp32" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/esp_hw_support/port/esp32/private_include" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/heap/include" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/log/include" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/lwip/include/apps" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/lwip/include/apps/sntp" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/lwip/lwip/src/include" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/lwip/port/esp32/include" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/lwip/port/esp32/include/arch" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/soc/include" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/soc/esp32" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/soc/esp32/include" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/hal/esp32/include" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/hal/include" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/hal/platform_port/include" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/esp_rom/include" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/esp_rom/include/esp32" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/esp_rom/esp32" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/esp_common/include" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/esp_system/include" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/esp_system/port/soc" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/esp_system/port/public_compat" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/esp32/include" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/xtensa/include" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/xtensa/esp32/include" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/driver/include" "-IZ:\\Users\\Jonathan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.17/tools/sdk/esp32/include/ [... cut ...]
In member function 'void SMLayerScrolling<RGB, optionFlags>::update(const char*) [with RGB = rgb24; unsigned int optionFlags = 0]',
    inlined from 'void loop()' at Z:\Users\Jonathan\Desktop\Projets\libraries\SmartMatrix\examples\FeatureDemo\FeatureDemo.ino:931:30:
z:\Users\Jonathan\Desktop\Projets\libraries\SmartMatrix\src/Layer_Scrolling_Impl.h:248:12: warning: 'char* strncpy(char*, const char*, size_t)' output truncated before terminating nul copying 27 bytes from a string of the same length [-Wstringop-truncation]
     strncpy(text, (const char *)inputtext, length);
     ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~`

davemaster, 2024-08-26:
> Finally I managed to compile the example provided in the library :) As proposed by @davemaster I downgraded to esp32 => 2.0.17 However, I have some Warnings that I will put for you below. [...]

Can You share Your output (in panel(s))? Thanks in advance

davemaster, 2024-08-26:
I will try with the 3.0.4 this night, later... keep in touch

HorstBaerbel, 2024-08-26:
> Can You share Your output (in panel(s))? Thanks in advance

Not right now, but the error is in __Impl.h_ files and always related to `gpio_pad_select_gpio`, as seen in the top posts

beinnlora, 2025-02-05:
For me, on ESP32-WROOM-DA board

- ESP board manager v 3.1.1 
- SmartMatrix library 4.0.3
- SmartMatrix GFX library 1.2.0
- Using MatrixHardware_ESP32_V0.h

add includes to headers depreciated between ESP board versions:

src/MatrixEsp32Hub75Refresh_Impl.h: 
`#include "esp32/rom/gpio.h"`

src/esp32_i2s_parallel.c: 
```
#include "soc/gpio_periph.h"
#include "hal/gpio_types.h"
```
the example 'MultipleTextLayers.ino" now compiles for me but unable to currently test on hardware